mudita/MuditaOS #3214: How to port to Raspberry Pi Zero (ZeroPhone)?

niutech, 2021-11-10:
#  📚 Missing or unclear documentation

## 📝 Description

Please provide documentation on how to port MuditaOS to other platforms such as Raspberry Pi Zero (BCM2835), which powers [ZeroPhone](https://hackaday.io/project/19035-zerophone-a-raspberry-pi-smartphone) with e-paper display (e.g. [Waveshare](https://www.waveshare.com/5.83inch-e-paper-hat.htm)).

## 📝 Describe the solution you'd like

There is a port of [FreeRTOS for BCM2835](https://github.com/varghes/Raspberry_FreeRTOS) and [another one](https://github.com/leodido99/RaspberryPi1-FreeRTOSv9.0.0), but maybe it would be easier to start off with the Linux port.

As an open source project, the OS shouldn't be tied to a single board. Thanks!

pholat, 2021-11-22:
Broadly speaking: it depends. Just running MuditaOS on RPI Zero shouldn't be too hard.
1. cross-compile the Linux target
2. run `./PurePhone` application on raspberry pi 0 target.
( There is a catch with it - we do not have any publicly available font so it will fail miserably at the start)
This would be a very good beginning.

Having the app compiled it shall run as it is. It won't be too functional though and very energy inefficient - due to the limitations of FreeRTOS emulation. If you connect the Quectel EC25 modem via USB - then it shall be not too hard to hack it to be a usable phone.
To make it properly energy efficient it would require either dropping the FreeRTOS emulation layer and switching to POSIX threads and pipes, or a better FreeRTOS wrapper. The first would be harder, but shall prove to be better in the long run. After that, it would require proper power profiling and power management adaptation.

Saying all of that, while striving to meet our goal which was a usable product it may not be the best-layered software, but an experienced user should be able to hack it to his heart content :) I would strongly encourage you to give the first step a try!

niutech, 2021-11-22:
Thanks for an explanation. As for Linux, how to install the missing font?
As for FreeRTOS, will MuditaOS on any hardware which is supported by FreeRTOS? Provided that RPi Zero has a usable port of FreeRTOS, will MuditaOS work out of the box?

pholat, 2021-11-23:
as for missing font - the issue is we do not have proper non-OSS distribution. The best way would be to provide an alternative font set. It's possible, but it requires a bit of work and as we are closing to release date - the time is at essence.
As for FreeRTOS - yes, it will run on any FreeRTOS supported platform, but you will have to provide BSP (board support layer) for each.
Provided that RPi Zero has a usable port and there is a BSP layer provided - it should work out of the box. 

The very first step will be: adding a new platform. Right now we have compilation for rt1051 or Linux, so you will have to add `TARGET` rpi0 and `CMAKE_TOOLCHAIN_FILE` - please look into `./configure.sh` 

In CMAKE_TOOLCHAIN_FILE you will have a compiler set for the Broadcom chip. Then it's pretty straightforward - try to compile and add the required BSP 
[image]

niutech, 2021-11-23:
Thanks again. As for the font, aren;t you using DejaVu Sans by default? It is based on [permissive and public domain license](https://dejavu-fonts.github.io/License.html), so can't you just bundle it? And by providing an alternative font, do you mean https://github.com/mudita/MuditaOS/issues/3216#issuecomment-975510308?

pholat, 2021-11-24:
Yes, DejaVu Sans is a default font - and with a reason, it has all the glyphs we needed :) Providing DejaVu is actually a pretty simple matter - but we shall plan it properly so that it won't be only added, but also it will be properly usable, documented etc. and that will take a bit of time.

Please let me know if you happen to configure a new target :crossed_fingers:

pholat, 2022-01-05:
No further comments, issue closed. Please reopen if required :)

closed 2022-01-05 by pholat