mudita/MuditaOS #3216: Missing repos?

brinlyau, 2021-11-12:
#  📚 Missing or unclear documentation

## 📝 Description

Are you missing the assets repo? (seems the script fails to download?) also, are you planning on opensourcing ecoboot

## 📝 Describe the solution you'd like

open sourcing the assets and ecoboot :)

wojtekidd, 2021-11-12:
Unfortunately, some of our assets are proprietary hence we can't fully open-source them 😞 
Regarding `ecoboot` - we'll discuss that internally and will let you know.

brinlyau, 2021-11-12:
Can you include them as a proprietary binary that we can link against? (or
support including them if we manage to extract them from the device ;) )

wojtekidd, 2021-11-12:
The first way seems more legit 😛 Could you please pos any good examples of projects that do this?

brinlyau, 2021-11-12:
Could you explain the issue in detail on which kind/type of binaries can't be shared and the actual restrictions (is it a restriction stopping you from posting source or distributing at all?) - so I can propose a good example (if it's the Luts.bin, that's one thing - I totally understand why you can't redistribute it), but the other files in the assets folder seem to be fonts and audio and translations.json)

I don't see how the project is meant to be built without them - it seems to error out.

I have a few ideas :-)

Is there an IRC/xmpp channel or discord/slack for community devs for a more real time interface for chatting?

brinlyau, 2021-11-15:
Any chance you could jump on IRC or the community discord so we could have a real time chat about this (in private if needed) (and you could share transcript internally about this)? -

geekoftheweek, 2021-11-17:
I also attempted to build the repo and failed on the asset-downloading step.  A workaround would be greatly appreciated.

brinlyau, 2021-11-17:
https://discord.com/invite/SjyV3Vjz @geekoftheweek maybe join and we can discuss this :)

pholat, 2021-11-22:
sorry for the late response :) 

To just compile you can empty the products `assets.json` file:
```
cat << EOF >> products/PurePhone/assets.json
{
    "assets": []
}
EOF
```
You will be able to compile just fine, but you won't have any fonts available. It's not a very good solution, but it's a starter.

Now. To just make the simulator work we can:
* provide open default font, we are actually using `./fonts/pure/dejavu_sans_bold_27.mpf` so it shouldn't be a problem
* distribute it in any repo really, just change assets.json to:
```
{
    "assets": [
        {
            "name": "./fonts/pure/dejavu_sans_bold_27.mpf",
            "output":"image/assets/fonts/dejavu_sans_bold_27.mpf"
        }
    ]
}
```
and change cmake: `cmake/modules/DownloadAsset.cmake`
line 30: ` --repository MuditaOSAssets` to the available repo with this `mpf` file.
It won't be pretty - but it will enable to build simulator image and run it. The application will use this font for everything as the default font :)  

To make it fully configurable, and i.e. use any font you wish with the app, we would have to extend efforts a bit, the easiest way I could see is:
- providing config file (flat JSON would be probably fastest to add) where we have `font-constant-name-in-app : font to use` mappings
- load it at the start of the app
- generate required fonts with https://github.com/mudita/fontbuilder
- add an option to download_assets.py to take local files.

Answering the question of - why these files are not available to the public, and talking just from my imperfect knowledge on the subject - we can distribute these files to be loaded in runtime by the software, but we can't distribute them as it is.

wojtekidd, 2021-11-22:
Some parts of the software are "unfortunately"  proprietary - you can keep track of them by reading the [LICENSE.md](https://github.com/mudita/MuditaOS/blob/master/LICENSE.md) + the font we use across MuditaOS is a proprietary one and needs buying a license when used for commercial purposes.

brinlyau, 2021-11-22:
Hey Adam and Wojtek.

to be clear, I did manage to get the simulator built (and running), and I
think I even got a buildable image working for the rt1051 platform
(required a lot to makefile hacking on a private branch) too - I don’t have
a device yet, but I compared my image with my official image and it seems
the same.

anyway; my request is option D “add an option to download_assets.py to take
local files”.

I have obtained my assets (fonts/Luts bin) from elsewhere (that I’m not
gonna disclose, and I have a private GitHub assets repo, as well as zip in
my downloads folder) but I’d like to use them in a build locally via a
option in a script (that I obviously can’t distribute) without maintaining
a seperate fork with seperate history..

I currently have them just straight inside a  repo as one initial commit;
because I’m not sure the best way to match your format with the GitHub
api/releases etc as I don’t know the structure of your _Mudita/Assets_ repo
;)  - if you can’t disclose that, then I’d love if you could just add
support for a local folder which assets could be imported from (we already
have gitcommit books to prevent stuff from being committed).

Brinly.

pholat, 2021-11-23:
Could you please check:
`/configure.sh pure rt1051 relWithDebInfo -DMUDITA_ASSETS_JSON=assets.json -DMUDITA_ASSETS_DL_TYPE=local`

Here: 
https://github.com/mudita/MuditaOS/pull/3309/files

You will just have to:
- provide what `assets.json` to use
- set `local` dl type
- list all files you want to copy in this file. I assume this would be all fonts you want to copy as in example `assets.json` in the root catalog.

Will it suffice? If so I could possibly update it in the meantime so that it would be somehow acceptable :)

brinlyau, 2021-11-23:
Perfect, that's basically what I was gonna propose, but I wasn't sure on what the best way for mainlining a patch like that was. I haven't tested yet, due to travelling, but if I get some time, I'll spin up a VPS on the weekend to try build :)

brinlyau, 2021-11-24:
https://gist.github.com/brinlyau/6596bf097ea116f7587a63faae555ae5

Picked those commits in [PR:,](https://github.com/mudita/MuditaOS/pull/3309) and added lol.txt in /tmp/ but it still fails to build. What did I miss? (I think the python script is broken)

pholat, 2021-11-24:
@brinlyau there are two downloads 1st is for assets and 2nd is for (assets) bootloader & loader which we download from selected releases.

Default target creates an image by default. For this case, you can build `PurePhone.hex-target` instead. It's pretty much useless without a phone though (we do not have any VM config for that :) ). Consider building for Linux and running simulator.

brinlyau, 2021-11-24:
I've been using the Linux simulator for about two weeks (using a patched up cmake), but i want to run on the real device since I have some patches (a bunch of my background is embedded/platform/bsp work) that are impractical with the emulator when mine arrives :) 

Are you planning on opensourcing the ecoboot binary (or even posting it in binary form on a publicly downloadable link) - I loaded the binary into IDA, and I could probably reimplement it, but I'd rather not if I don't have to? 

If anybody is curious: these are the assets that I have (BYO as I cannot supply anybody them), 
https://gist.github.com/brinlyau/080ed55b1bd092fc00b14f45ec4671ca

pholat, 2021-11-25:
That's great :) 

Opening up is a process, as MuditaOS is OSS, PureUpdater is next in line and should be available shortly, then I suppose ecoboot will follow up. While the ecoboot is signed, the signature is used for authentication only. 

So the very first need you will have is to load binary on the phone. I do not know if NPI has a JTAG socket soldered in - but I guess this would be the easiest way to load the software for tests. There are a few other ways to load the software, this is just the simplest and least intrusive one. 

Please tell if delivering the proposed draft will be enough to close this issue  :crossed_fingers: 

ps. I know that @SP2FET and @paweljonskim were recently announced as community contact developers on the forum - if you happen to send some ideas, patches, or even remarks what would you expect from us as an OSS developer - I'm sure it won't be unnoticed.

brinlyau, 2021-11-25:
Perfect. I think we can close this issue now with the proposed draft
patches.

Thanks for your work on this.  Super cool to have a open source phoneOS
that’s not a smartphone :)

brinlyau, 2021-12-08:
Closing this issue, but might open up another one to explain the workaround :)

closed 2021-12-08 by brinlyau