mudita/MuditaOS #3337: Minimum requirements

niutech, 2021-11-27:
Please provide minimum requirements to run MuditaOS on a physical board with FreeRTOS. Will ESP32 with 8MB PSRAM be enough to boot it?

wojtekrzepecki, 2021-12-02:
Most of the constraints are about memory usage. I.MX RT1051 microcontroller have 512KB of on-chip RAM which is utilized  for threads stacks, FreeRTOS heap etc.. Mudita Pure is equipped with 64MB of SDRAM. Here you can find memory utilization of the recent build:
[image]
Mudita Harmony (in code: Bell) has 16MB of SDRAM and it's totally sufficient but the binary is much smaller (3,5MB).

You have to be aware that design was not optimized for low memory footprint, e.g. all images are loaded into heap at the system initialization. With some tricks 8MB of RAM may be enough to fit Harmony build. Running Pure Phone may be more problematic.

niutech, 2021-12-02:
Thank you. How about processing power? Dual-core 240 MHz CPU should be enough to boot it, or do I need much more?

wojtekrzepecki, 2021-12-02:
We are running Pure with top 528MHz CPU frequency, which is decreased most of the time according to cpu load dynamically. 240MHz should be enough.

wojtekrzepecki, 2021-12-14:
@niutech Is this information sufficient for you? If yes, please close the issue :slightly_smiling_face:

niutech, 2021-12-15:
Yes, thank you.

closed 2021-12-15 by niutech