mudita/MuditaOS #3402: Phone crashed

snan, 2021-12-11:
After updating the OS to 1.1.15, I've had the phone crash twice.

First, I previously already had Ukulele selected as my ring tone, then I connected to Mudita Center and updated the OS, and all of the following is with the Mudita Pure connected to the computer and Mudita Center open but not connected. 

Then after the update (and with the wire still plugged in),
go to settings -> apps -> phones and then the list of ring tones.
With the cursor on Ukulele (and there's a check mark there), hold the back key for a while, then let go. The phone will become unresponsive and then restart.

SP2FET, 2021-12-15:
Hi!
Indeed, the phone freezes when you long-press the back key in the scenario that you described, thanks for pointing it out. I'll pass it further to devs team

SP2FET, 2021-12-15:
I've created a bug on Jira, so you'll see in the PR section it's progress once it will be started: EGD-8104

closed 2021-12-20 by PrzeBrudny

SP2FET, 2021-12-23:
@snan the issue has been fixed :)