mudita/MuditaOS #3404: More beautiful meditation countdown

snan, 2021-12-12:
The meditation time is on a proportional font so as the seconds tick down, the entire time stamp jitters. If instead each glyph was fixed in space only the relevant number would move. It could still be relatively proportional, i.e. a thin `:` and so on. This would look awesome on e-ink ♥

+ also add a 20 minut schedule ♥♥

pholat, 2022-01-05:
Actually whole timer should jitter as i.e. 1 and 0 have different width and inter font parameters. Saying that - wouldn't at best be the use od monospaced font in this particular example?

snan, 2022-01-05:
As long as it's only the last digit that's changing, it wouldn't jitter. And going from 50 to 49, for example, both digits are changing at the same time. It's never the case that a non LSD changes on its own.

snan, 2022-01-05:
So a left-justified (a.k.a. "ragged right") timer doesn't jitter.

pholat, 2022-01-12:
@snan non-monospaced fonts have two parameters that should differ for different glyph combinations: this is [kerning](https://www.wikiwand.com/en/Kerning) and glyph width. 
Consider glyphs for `1` and `2` - both have different widths. Now when you write 50 times in i.e. LibreOffice with different fonts these would have different widths:
[image]
This is why I said that the whole timer should jitter. I felt that you suggested forcibly "holding" each character (in range of glyphs) and that actually makes non-monospaced font behave as a monospaced one. This is why I suggested monospaced font. As it has the same width and no kerning for each character, it's just the simplest approach. 

Saing all of that you would be pretty happy if it didn't jitter :)

mjan-dev, 2022-03-07:
@snan FYI, we are planning to introduce monospaced font for use with timers across the system. 
I'm closing this issue for now. Thanks for the suggestion :+1:

snan, 2022-03-07:
Wonderful ♥
Aesthetic touches & polish ftw ♥♥♥

snan, 2022-03-07:
> @snan I felt that you suggested forcibly "holding" each character (in range of glyphs) and that actually makes non-monospaced font behave as a monospaced one.

I missed this reply.

I know how widths and kerns work.

A monospaced font timer doesn't jitter, but neither does a ragged-right (a.k.a. left-aligned) proportional font timer. (Because a digit can only change when all of the digits to the right of it changes, that's just math.) My suggestion was to make the minutes and seconds both be separately left-aligned (a.k.a. ragged right) and keep the `:` character steady between them.

cha
[image]