mudita/MuditaOS #3408: Hangups, reboots, bluetooth, slow responding and battery drainage

czalaviek, 2021-12-13:
# 🐛 Bug report

## 📝 Description

I'm fresh Pure user, since 3 days and after 1'st connection with Mudita Center I've upgraded firmware to 1.1.5. It was bad decision that brought me hangups (often +reboot), slow responding, battery drainage and troubleshooting with bluetooth.  I've imported about nine hundreds of contacts from my Google account.

## 👍 Expected behavior

1. Battery should work longer than one day after full charging.
2. Bluetooth connection should always work with both a car audio system and wireless headphones
3. Multitasking should let user the same time write a message or modify contacts and receive a message without any GUI hangs
4. After every, soft or hard, reboot, phone should ask for PIN of SIM card.

## 👎 Current behavior

1. Battery allows only for max 2 days of work (strength of my mobile network is 2 from 4 bars). Back isn't not a bit warmer than front cover.
2. Bluetooth connection with my Toyota '11 car audio system was possible only once. The sound was terrible with annoying crackles. 
3. When I was editing any contact, I've received SMS. Unfortunately phone has stopped responding. There was only possible to turn on keyboard backlight by pressing any key. After a while phone rebooted. I met the same behaviour many times during viewing messages.
4. After every reboot, phone's SIM card is inactive as for lack of prompt for PIN.

## 🔬 Minimal reproduction

Import lots contacts (ex. 900)
Only one SIM-card in 1'st port
Polish locales
No other major customisations.

## 🔍 Provide logs

Automatically sent by Mudita Center.

## 🌍 Your operating system and MuditaOS version
1.1.5

## 🤔 Anything else relevant?
If u need I can continue in polish :)

SP2FET, 2021-12-16:
Hi!
I'm sorry to hear that you had so many problems with your Pure.
Let me answer the Bluetooth issue, others will be addressed by my colleagues:

You're right, that car connection could be a problem - we did internal tests and we're working on solving the bug which causes problems with car BT systems.

Let me continue in English for the sake of other community devs ;)

wojtekrzepecki, 2021-12-16:
1. Battery life depends mostly on cellular network coverage and how much the phone is actually used, especially for calls. Also Bluetooth module draws substantial amount of power if used. When phone is idle and the cellular network signal is strong and stable phone should work about 5 days on single charge. If you are using Bluetooth often and the network signal is usually not constant, it is likely that battery may be drained much faster, as you observed.

mjan-dev, 2021-12-16:
Let me answer remaining points.
3,4. Those are clearly bugs and an annoying ones if I may add (especially the third one). Furthermore, the SIM issue may influence battery life. We are going to address this as a high priority issues and hopefully those will be solved quickly. Thanks for reporting it and for actively contributing to improving Pure user experience!

mjan-dev, 2021-12-16:
Bugs tickets have been created to address reported issues 3 and 4. This issue will be updated immediately with progress. 
[EGD-8116](https://appnroll.atlassian.net/browse/EGD-8116) 
[EGD-8115](https://appnroll.atlassian.net/browse/EGD-8115)

czalaviek, 2021-12-16:
Ad 1.
I live 400 m to the nearest cellular operator's BTS (full LTE power 5/5 in my smartphone) and my work office is located in voivodeship city center. There is n no reason on BTS. Furthermore, I don't use Bluetooth, because it's unusable (desc. in Ad 2)

Ad 2.
It's very hard to make any pairing with car audio (Toyota and Renault). After several attempts, Bluetooth connection was successful, in Pure and a car. Unfortunately, Pure is disconnecting after few seconds, again and again. I's very nervous - Actually, I've to forget about calling by handsfree in my cars.
By the way: Bluetooth should be automatically turned off after fixed minutest of no use.

Ad 3. & 4.
I look forward to it in a new public-release of firmware.

PS. Maybe firmware isn't bugs-free, but my experience shows that hardware is "ill", too.

mjan-dev, 2021-12-17:
@czalaviek Thanks for the logs. Those have been added to proper issues and will be extremely helpful in solving them.

czalaviek, 2021-12-18:
Thx. I'm waiting for results in next public release of firmware. It would be nice to see it as fast as it's possible

SP2FET, 2022-04-25:
Hi,
Ad 1. First battery improvements have been already introduced in the 1.2.0 release, next major improvements will be included in the following updates. (Those are already ready for release)

Ad 2. Bluetooth car audio should be fixed in the coming release, so stay tuned ;)

Ad 3,4. Those bugs should be already fixed in the current release.