mudita/MuditaOS #3418: Concatenated strings cause ambiguity for some languages

bitigchi, 2021-12-15:
# Concatenated strings cause ambiguities for some languages

## 📝 Description

I have been translating MuditaOS into Turkish. However, I am having trouble translating some strings, because they are not whole strings but rather parts of it that changes according to context.

Translating these strings may not issue great problem for European languages and their mostly similar word order, but for different language families like Turkish, where the word order is complete opposite of a regular Indo-European language, this consists a great problem.

Some problematic strings:

- `"app_desktop_info_mmi_common_no_message"`: "not" follows word in Turkish, so I have no idea figuring out what is "not". Valid for the whole string group here, but to a less extent. Ideally it should not be used at all.
- Below group:
```json
  "app_bellmain_home_screen_bottom_desc": "Sonraki çalar saat zamanı:",
   "app_bellmain_home_screen_bottom_desc_in"
   "app_bellmain_home_screen_bottom_desc_and"
   "app_bellmain_home_screen_bottom_desc_less_than"
```
Currently this group of strings are translated very fuzzy and it barely makes the context understandable.

## 👍 Expected behavior

MuditaOS strings should be more l10n friendly. Variables and numbered variables (in case there are more than one in one string) to accommodate languages which has a different word order. Concatenated strings are generally frowned upon for l10n purposes, and should be avoided altogether.

pholat, 2021-12-17:
Thanks for reaching out. Please right me if I understood you wrongly: In this particular case there is a piece of code that acts as a generator and glues a few strings together, whereas for translation purposes it should just return one of 5 different translations.

So i.e. instead returning (simplified):
`app_bellmain_home_screen_bottom_desc` + `app_bellmain_home_screen_bottom_desc` + ... `something` 
it should be:
`app_bellmain_home_screen_bottom_desc_alarm_in_24_h` 
(and 4 other strings which would be translator friendly after)

Would you be able to help us out here? I'm asking only because this would be most probably the fastest way to address the issue. If not - after confirmation we will tackle it as a standard bug in the code internally :)

bitigchi, 2021-12-17:
Yes, correct. I've prepared an initial fix which broke down the strings a bit further, which allows me to relocate the verb, but I am not sure how to put variables into translatable strings like `$VARIABLE$` to make it into a single string. Otherwise it works.

pholat, 2022-01-03:
do you have it somewhere? please link it here, we would be more than happy to help you get through the fix admission process ( as well as help to implement it)

pholat, 2022-02-15:
@bitigchi hi, are you still interested in the change? I believe that with some small changes to your commit: https://github.com/mudita/MuditaOS/issues/3418 you would be able to PR and we could push this issue through and close it :)

bitigchi, 2022-02-15:
@pholat, I lost the original patch, but I'll have a fresh look with a bit more study of the source code.

Lefucjusz, 2022-07-18:
Closing the issue due to lack of response, if the problem persists please reopen the issue.

closed 2022-07-18 by Lefucjusz