mudita/MuditaOS #3421: Battery depleting quickly with idle

snan, 2021-12-16:
So the battery meter, at least the visual indication of the battery, is depleting pretty quickly when idle. Too many writes to the e-ink screen or some other issue?

I'm used to 3G phones, even dumbphones, depleting quicly when actually talking, but the point of the Mudita is to not use it (non-addictive phone) and still be reachable.

wojtekrzepecki, 2021-12-16:
Hello @snan ,

Thank you for reporting the issue. I have some questions for clarification:
1. Was SIM card in the phone all the time?
2. What was the cellular signal strength at the time?
Battery life is mostly influenced by instabilities in cellular network signal. When signal is weak/unstable the GSM module has to transmit with significantly higher power. In stable network conditions phone should work for about 5 days in idle on single charge (release 1.1.5).

snan, 2021-12-16:
Yes. Slider on first two modes (not airplane).
Signal is top left, right? I get two bars.

wojtekrzepecki, 2021-12-16:
Could you provide more data to diagnose the issue? There is an option to do this by Mudita Center. While Pure is connected by USB to PC, please click "contact support" icon. Following form should appear and the logs from the phone will be attached there. Please also attach there this Github issue number :slightly_smiling_face: 
[image]

snan, 2021-12-16:
Is there a way to see those logs before sending them? So I can verify that my location, phone number etc isn't transmitted.

wojtekrzepecki, 2021-12-16:
Yes, you can check them in the temporary folder used by Mudita Center:
- Windows: `C:\Users\<username>\AppData\Roaming\@mudita\mudita-center-app\logs`
- Linux: `~/.config/@mudita/mudita-center-app/logs`
- macOS: `~/Library/Application Support/@mudita/mudita-center-app/logs`
We have made lots of effort to be sure that no sensitive data is stored in logs. Anyway, we encourage you to check them out :wink:

snan, 2021-12-16:
Oh, those logs are pretty great, I used them the other day to fix the [permissions issue](https://forum.mudita.com/t/mudita-center-on-debian/3382).

I'll go grab those logs the next time I can connect to a computer.

SP2FET, 2021-12-17:
FYI bug is reported under EGD-8120 ticket ;)

SP2FET, 2022-01-31:
Hi, let me inform you that we found the issue and we're on track to fix it as quickly as possible ;)

mjan-dev, 2022-03-07:
Possibly, the main reason of this issue has been addressed and will be included in newest 1.2.0 release. 
Furthermore, we are in progress with work on major improvement of battery life as whole. Closing the issue.

closed 2022-03-07 by mjan-dev

snan, 2022-04-24:
Battery seems much better (just, IDK, feels that way without measuring too carefully).