mudita/MuditaOS #3424: Possible memory leaks between received messages

czalaviek, 2021-12-16:
I observed today weird behaviour of messaging app. 1'st I've received message from my banking service (lack of number, only name like "PKO BP"). Message was cutted-off, only about 10 letters left. After a while, I've received next message, from my friend. To my surprise message was concatenated from rest of banking msg and original friend's text.

mjan-dev, 2021-12-17:
Hi czalaviek!
Sorry to hear about the issue. Good thing is we are already working on it and we had it in backlog :) 
Here is some tracking number to see when it will appear on PR section. [EGD-6991](https://appnroll.atlassian.net/browse/EGD-6991)

SP2FET, 2022-01-31:
Hi, 
Let me inform you that this issue has been fixed and will be included in the next release.

closed 2022-01-31 by SP2FET

czalaviek, 2022-01-31:
Great news!!!