mudita/MuditaOS #3427: Sounds bad

snan, 2021-12-18:
I was asked on the forum to file an issue for the speaker's audio quality but I'm not sure that there's a lot you can do about it in software.

Drivers?
Gain / eq?
The speaker sounds overdriven and clippy.
[Here is a comparison with spectrogram.](https://forum.mudita.com/t/reviews-of-mudita-pure/3375/42)

mjan-dev, 2022-01-03:
Hi snan!
Thanks for reporting this issue here as well as on forum. We are already working on it, recording have been made and quick adjustments to equalizers have been proposed and are currently review internally. Of course, changes to sound quality are limited by hardware and mostly enclosure design, however some tweaks are possible to be introduced to loud speaker sound and especially to the earspeaker sound. We aim to release tweaks in this area in the nearest release version and work on further improvement after that.

LizGalanis, 2022-04-10:
Glad to see this is reported already! I'm finding that the ear speaker is really distorted. When volume is up high it's basically unusable, at lower volume it's okay but still a bit distorted.

SP2FET, 2022-05-23:
hi @snan and @LizGalanis - as you can see, we managed to take some serious measurements and find some tweaks that improved the call quality. Probably more info in detail will be available soon on the forum ;)

closed 2022-05-23 by SP2FET

LizGalanis, 2022-07-06:
> 

Hi Sp2fet, unfortunately my Mudita phone is no longer recognising any SIM cards and I cannot make or receive phone calls. Thanks for making these improvements, but they are not relevant to me due to the phone having no call functionality.