mudita/MuditaOS #3428: Datestamp of received messages

czalaviek, 2021-12-18:
# 🚀 Datestamp of received messages

## 📝 Description

Actually every message is described only by short hh:mm stamp, even if they are older than today's. 

## 📝 Describe the solution you'd like

Time stamp of old messages should be built just like: a) older than 7 days: [MONTH],DD, HH:MM; b) younger: [DAY], HH:MM; c) another year: YYYY-MM-DD, HH:MM

## 📝 Describe alternatives you've considered

none

mjan-dev, 2022-01-05:
Thanks for creating the issue and for your contribution.
It is a valid remark and we will plan and implement change in this field and soon as possible.

GravisZro, 2022-03-12:
FYI, the issue is [here](https://github.com/mudita/MuditaOS/blob/dfde2e1b0a61212f4fe25c7a5afe931786a3c2c3/module-apps/application-messages/widgets/SMSOutputWidget.cpp#L131)
```cpp
       timeLabel->setText(*TimestampFactory().createTimestamp(TimestampType::Time, timestamp));
```

GravisZro, 2022-05-02:
# 🚀 Feature request / Proposed solution

## 📝 Description

The text message conversation log should have a divider between the days.  This would solve issue #3428.

## 📝 Describe the solution you'd like

The divider should include the date for the messages above with an upward arrow as well as the date for the messages below with a downward arrow.

## 📝 Describe alternatives you've considered

Putting the full datetime stamp on a message would work but it wouldn't be as good if it only shows when highlighted.

Lefucjusz, 2022-09-19:
@czalaviek @GravisZro thanks for reporting the issue. We found another bug in SMS date handling, https://appnroll.atlassian.net/browse/MOS-722, related with the problem you've reported, the feature you've requested will be added during fixing of the bug and will be relased in one of the next MuditaOS releases.

Lefucjusz, 2023-02-10:
@czalaviek @GravisZro the exact issue you've described has been reported under MOS-893 ticket and both with the aforementioned MOS-722 they're already resolved, the fixes will be present in one of the next MuditaOS releases. After the change:

- for the today's messages only hour is shown;
- for the yesterday's message "yesterday" and hour is shown;
- for the messages from the last week day of the week ("Mon"-"Sun") and hour is shown;
- for the older messages short form of date is shown ("DD.MM") together with hour;
- for the messages from the previous year full date is shown ("DD.MM.YY") together with hour.

closed 2023-02-10 by Lefucjusz