mudita/MuditaOS #3448: Deleting first char while editing messages

czalaviek, 2021-12-23:
# 🐛 Bug report

## 📝 Description

Pressing del key [#], when cursor is located on 1'st position, removes 1'st right char.

## 👍 Expected behavior

Del key should do nothing when cursor is located on the start position.

## 👎 Current behavior

Pressing del key [#], when cursor is located on 1'st position, removes 1'st right char.

## 🔬 Minimal reproduction

Situation is so simple and it doesn't need long description :)

## 🔍 Provide logs

<pre><code>

</code></pre>

## 🌍 Your operating system and MuditaOS version

## 🤔 Anything else relevant?

czalaviek, 2021-12-26:
IT looks that error od harder, because the same situation is any textbox created from the same stereotype, ex. fields in contacts.

pholat, 2022-01-03:
This is dependent on how the UI text widget is used in particular cases. Generally speaking, behavior when we are with cursor prior to any other character, we press delete, we remove character after (not prior as there is none) was left as a feature in code as it was proven handy - it's hard to press `backspace` by accident on the phone keyboard too. Therefore, it didn't feel as it would bother anyone.
Could you please point out specific places where it's bothering you and why?

czalaviek, 2022-01-03:
Cases:
1) Editing new message - I wrote "Horse", but I wanted to write "Morse". To correct the word, I'm moving cursor to position between "H" and "o", next move is hitting "#" as Backspace. Unfortunately I pressed it twice. Result is deleted char "H" but "o" too. It's unwanted behaviour, because needs to write two chars, now.
The same behaviour is while editing Name, Surname etc. in contacts.

Pure's keyboard isn't ASCII-keyboard device so every possibility to avoid unneeded keystrokes is good decision.

mjan-dev, 2022-03-07:
The backspace behavior here is working as designed. We will gladly consider your suggestion, especially if more users will report similar feedback. For now I'm closing this issue.

closed 2022-03-07 by mjan-dev