mudita/MuditaOS #3454: Unread messages and missed calls on lock screen - inconvenience

czalaviek, 2022-01-04:
# 🚀 New messages and missed calls seen before auto-lock screen should be marked as read

## 📝 Description

Current behavior: when user has seen every missed call or unread messages. Next, he put the phone down without closing contacts- or messages-app. OS auto-locked the screen after a fixed time, but missed calls or unread messages are still on the lock-screen.

## 📝 Describe the solution you'd like

Messages and missed calls should be marked as read/seen after opening, not after closing messages screen or calls log. This way lock-screen will be free from read messages and seen missed calls.

mjan-dev, 2022-01-13:
Thanks for reporting. It is indeed a bug and I created an issue for it with id: EGD-8163. It can be tracked as soon as it appears in PR.

mjan-dev, 2022-03-07:
Closing. It is fixed and will be included in 1.2.0 release.

closed 2022-03-07 by mjan-dev