mudita/MuditaOS #3485: Rejecting calls

snan, 2022-01-16:
Friend called me, I was busy, so I hit the rightmost bar nav key. From my perspective, my friend called again, and I answered. And in the call log, I have two calls from 16:36, one with an x and one with a checkmark.

From my friend's perspective, he called once, and heard one "call signal beep", then a few moments of silence, then more "call signal beeps" (with "lower quality" according to him), and in his phone's call log, he has one call at 16:36 marked green (because I answered).

snan, 2022-01-18:
And then, next night, same friend called again (from same phone, some iPhone I guess), and I had sound off, so didn't even reject (or hit the right bar key), just noticed later two missed calls both same timestamp, and he told me today that on his end it was only one call. He reports a long pause before the "call signal beeps" that time.

pholat, 2022-01-20:
First of all: thanks! We are actually trying to reproduce it from the first time you posted but weren't able :) would you be so kind to tell:
- do you have voicemail enabled?
- how many +- contact do you have added? 
We would love to get a bit closer to the comparable environment to yours ( as normally when we load the phones for reproduction we have voicemail on and just a few contacts)

snan, 2022-01-20:
I had voice mail with my older dumbphone. Not sure how it works on the pure. I had eleven contacts added at the time.

mjan-dev, 2022-02-14:
@snan As we are having trouble reproducing some of your issues, we requested for direct contact in https://github.com/mudita/MuditaOS/issues/3553. We will take a further look if either your report further details or we come across the same issues as reported.

closed 2022-02-14 by mjan-dev

snan, 2022-02-14:
That request was just one week ago!