mudita/MuditaOS #3493: SMS mark read

snan, 2022-01-18:
When an SMS shows up in my open thread it is still marked as unread once I leave the thread.

SP2FET, 2022-01-20:
Hi, I've tested the case on the 1.1.5 release and the current `master` branch and can't reproduce it - the thread is marked read and there's no dot next to the thread. Could you please try to reproduce this?

snan, 2022-01-20:
I'm on 1.1.5 and it happens all the time:

The pure is open to a chat,  
a new message comes in,  
the new message is shown in the chat,

but then the pure doesn't know that the message is read. It shows up on the main (pre-menu) screen and there's a dot on messages and on the thread. Unless I go back in there.

SP2FET, 2022-01-20:
ah, I see, this happen when you have the thread opened while the message comes? I didn't understand it correctly for the first time. I'll recheck it

SP2FET, 2022-01-20:
alright, you're correct, I'll create a bug in Jira

SP2FET, 2022-01-20:
Thanks for noticing the bug, I've created a ticket for this: EGD-8194

czalaviek, 2022-01-21:
I thing the issue is similar to [my added earlier](https://github.com/mudita/MuditaOS/issues/3454 )

SP2FET, 2022-01-21:
well, indeed it's the same case - we'll merge the tickets then. thanks for pointing it out!

SP2FET, 2022-01-24:
duplicated: https://github.com/mudita/MuditaOS/issues/3454

closed 2022-01-24 by SP2FET