mudita/MuditaOS #3507: Wifi wireless connection to save data from a data plan and thus save money

dumblob, 2022-01-20:
# 🚀 Feature request

## 📝 Description

It seems impossible to connect to any wifi network nor to do any tethering over wifi.

## 📝 Describe the solution you'd like

I'd like a simple list of available wifi connections to choose from, then put password in and subsequently use the connection from applications (e.g. to write messages, write emails, VoIP, etc. - just about anything one can do now with an ordinary LTE data connection).

As a nice to have I'd like to have the opportunity to tether my LTE data plan over wifi to multiple devices (i.e. not just through one USB cable) - e.g. notebook & [Betrusted Precursor](https://www.crowdsupply.com/sutajio-kosagi/precursor/ ).

## 📝 Describe alternatives you've considered

Using a USB wifi dongle. But that's very cumbersome, unsupported by MuditaOS and more importantly highly energy inefficient.

pholat, 2022-01-21:
It's not possible, which was a deliberate design choice. Please mind that WiFi requires both specialized software and hardware and we have none of both :) It might be possible to have Bluetooth tethering - as we have Bluetooth classic capabilities, but it was not implemented.

dumblob, 2022-01-22:
>It might be possible to have Bluetooth tethering - as we have Bluetooth classic capabilities, but it was not implemented.

Is it planned to implement this functionality (Bluetooth tethering)?