mudita/MuditaOS #3553: Accepting calls

snan, 2022-02-06:
I got a phone call that I wanted to accept but I couldn't. I kept hitting "accept call" but the button did nothing. Then I finally hit reject, and went to the contact lists to call him. The call button again did nothing. I restarted the phone and tried to call from the call list. It says "put in a sim card" but I do have one in.

snan, 2022-02-07:
I think the sim card thing was just from having restarted so recently as opposed to a physical glitch.

SP2FET, 2022-02-07:
HI, 
We're sorry that you're experiencing new problems with your Pure, even more, that those are related to the cellular connection.
To be able to help you as much as we can we would like to contact you in a more "instant" way - maybe some sort of instant messaging? We could then try to investigate your cellular problems based on your individual device.
If you'd like to take part in such a conversation I'll leave here my email - we can then find the most suitable platform to talk, taking into account your preferences.
Mail: bartosz.cichocki@mudita.com

snan, 2022-02-14:
We can talk on IRC

snan, 2022-02-16:
It happened again today as I was accepting a delivery; I was getting called but I had no way of accepting the call. The left-most context key didn't work.
Also, I could see the number calling me, but looking in the call list later it just says "unknown" (frustrating since I couldn't call him back). Delivery guy later tried again and that time, I could answer, and in the call list his number is shown.

closed 2022-06-20 by Lefucjusz