mudita/MuditaOS #3554: Alarm can't stop when left selection key pressed

czalaviek, 2022-02-06:
# 🐛 Bug report

## 📝 Description

Alarm can't stop when left selection key pressed. It opens main menu. Switch off alarm in Alarms app brings no help - phone is still loud. No GUI alarm notification, now. Helps only turning off the phone - long press right selection key. Volume keys are disabled.

## 👍 Expected behavior

Left selection key should by disabled during alarm in progress

## 👎 Current behavior

Current Left selection key during alarm in progress opens main menu. Volume keys are disabled.

## 🔬 Minimal reproduction
1. Set any alarm
2. Press left selection key during alarm.
3. Try turn off alarm
4. Try press volume buttons.

## 🌍 Your operating system and MuditaOS version
1.1.6

pholat, 2022-02-07:
Hi, this should be fixed in the new release 1.1.7 which should be released in the following weeks. If there are no major issues found, I assume it would take about a week.

czalaviek, 2022-02-07:
👍

mjan-dev, 2022-03-07:
It is fixed and will be included in 1.2.0 release.

closed 2022-03-07 by mjan-dev