mudita/MuditaOS #3555: Missed calls in log still bold (flag "unseen")

czalaviek, 2022-02-06:
# 🐛 Bug report

## 📝 Description

I've reviewed all my missed calls in call log, but they are still bold

## 👍 Expected behavior

All call log position should be unbolded alter review.

## 👎 Current behavior

Reviewed all missed calls in call log are still bold

## 🌍 Your operating system and MuditaOS version
1.1.6

SP2FET, 2022-02-09:
Hi,
The issue indeed exists, it is probably related to the issue you mentioned before and we're on track to fix it!

SP2FET, 2022-02-10:
I've created a separate ticket for this bug:
https://appnroll.atlassian.net/browse/MOS-158

SP2FET, 2022-02-11:
Hi, I've got some info that this is not a bug, it's a feature. The bolded entries meant to be the missed calls - that's the way it's designed. If we'll got more complaints about this, it will be reworked then.

mjan-dev, 2022-02-14:
@czalaviek I'm closing this issue as it has been reviewed by the team and this behavior is indeed as designed. As @SP2FET mentioned, we will gladly redesign it if more complaints about this feature arises. Thanks for reporting.

closed 2022-02-14 by mjan-dev