mudita/MuditaOS #3596: Airplane mode grounded

snan, 2022-02-17:
I had the phone in Airplane Mode (I went and checked the settings, it wasn't in "periodically download messages" mode, although it might'e thought it was because that'd explain the following behavior): I got the message chime and saw that I had a missed call text message (from my telephony provider) and two other messages. The missed call was just nu, 13:11, same minute. I went to call the caller (since I didn't remember that I was in airplane mode) but it said "you're disconnected, connect to make a call" and then I slid the switch up and called him.

mjan-dev, 2022-03-07:
Hi snan!
Thanks for reporting. Is this issue recurring for you or was it only a one time occurence? 
Either way, could you please send report through Mudita Center so we can look through your logs. Much appreciated.

SP2FET, 2022-04-25:
Closing due to inactivity. If the issue appears again, please reopen the issue.

closed 2022-04-25 by SP2FET