mudita/MuditaOS #3597: Access request for PureUpdater

GravisZro, 2022-02-17:
# ⚠️ Access Request

## 📝 Description

I've followed the [Download Assets](https://github.com/mudita/MuditaOS/blob/master/doc/download_assets.md) guide for making the `download_asset.py` script work but after getting a token and configuring git, it's giving me a 404 still.  The instructions read 
> if not Then fix your token, or request access to PureUpdater

which leads me to request access to the `PureUpdater` repo because I've set git's config vars properly with a token with the "repos" section and subsections enabled.
```
$ git config user.githublogin && git config user.apitoken
GravisZro
[GITHUB ACCESS TOKEN STRING HERE]
```
## 📝 Describe the solution you'd like

Access to the `PureUpdater` repo.  If that is not possible then I would or a way to build the firmware and unit tests without assets.

## 📝 Describe alternatives you've considered

I've looked the docs and closed issues to look for a way to build without assets to no avail.

estevezluis1, 2022-04-08:
I'm not sure how access to PureUpdater will be granted, but in case it's manual, I would like to access to PureUpdater repo to build muditaOS

inconspicuouspilgrim, 2022-04-10:
Same here, not sure if one is required to reach out personally to the team for access

pholat, 2022-05-26:
Update: 
- you are able to build packages with community targets. Please see: `doc/build_targets.md`
- you are not able to download our proprietary software. We have MuditaOS as open-source, but the updater or bootloader is not open source yet. I've no information on when and if it will be opensourced.

Saying that - Updater is not part of the OS and is not open-sourced at the moment.

GravisZro, 2022-05-26:
@pholat 
The real question is when will I be able to build and load my own version of the OS because that's the whole point of open source.  Without being able to do this or having a serial debug link then I will be unable to determine the cause of the many "unreproducible" ills my phone suffers, like going indefinitely without being able to get a signal, or why my car believes the phone gave it the incorrect bluetooth password.