mudita/MuditaOS #3606: Checking with PVS-Studio static analyser

Stolyarrrov, 2022-02-22:
Hello,

I have found some weaknesses and bugs using PVS-Studio tool. PVS-Studio is a static code analyzer for C, C++,C# and Java: 

- Here is the [link](https://pvs-studio.com/en/blog/posts/cpp/0914/) to the first article about it
- Here is the [link](https://pvs-studio.com/en/blog/posts/cpp/0925/) to the secound 
- Open Source projects [can use](https://pvs-studio.com/en/blog/posts/0614/%C2%A0) PVS-Studio as free

Best regards,
Vladislav Stolyarov

Stolyarrrov, 2022-02-22:
Opening this issue is my typo, could you please delete it?

SP2FET, 2022-02-23:
Hi, thanks for the issue!
We already saw your posts and we're on the way to fixing it and probably integrating the tool with our CI flow :)
Here's the ticket in case you'd like to track the fixing in PRs: MOS-240

closed 2022-04-05 by paweljonskim