mudita/MuditaOS #3609: Chinese localization 🇨🇳

gabryatfendor, 2022-02-22:
A user on the mudita forum wanted to help with the Chinese localization but the whole Github process was too complicated, so he just shared with me the translated JSON.

gabryatfendor, 2022-02-22:
I created a PR, https://github.com/mudita/MuditaOS/pull/3611

c933103, 2022-04-14:
1. According to https://github.com/mudita/MuditaOS/blob/master/doc/i18n.md , only characters in https://github.com/mudita/fontbuilder/blob/master/charset.txt are currently supported in the OS. So all these Chinese characters wouldn't be displaying even after they are added, unless relevant font is made and added into the OS.
2. The added translation is only the Simplified Chinese, China version of the Chinese language. Additional localization for Traditional Chinese, for users in Hong Kong, Taiwan, and such, will be needed. Hence I don't think it's a good idea to just name the translation "Chinese".

toastal, 2022-05-19:
Oh that's a total bummer. I was looking at a way to get my hands on this device, but without Thai or Lao support, it would be unusable to me.