mudita/MuditaOS #3657: Incoming call while editing contact freezes and reboots Pure

rawzen, 2022-03-18:
1. Pure is set to DO NOT DISTURB mode. (The bug won't reproduce for me in CONNECTED mode.)
2. Enter editing mode of any contact.
3. Move focus to another field in the contact. (The bug won't reproduce for me if I don't move focus.)
4. Make a call to Pure.
5. The call does not come through.
6. Pure freezes, no buttons are responsive.
7. After a long press on the hang up button the phone reboots by itself.

SP2FET, 2022-03-21:
Hi, which firmware version are you using? The latest one (1.1.6)?

rawzen, 2022-03-21:
Yes. It is version 1.1.6.

mån 21 mars 2022 kl. 08:36 skrev Bartosz Cichocki ***@***.***

SP2FET, 2022-06-13:
Unfortunately, we can't reproduce it on the latest release. Closing right now, if the issue persists, please reopen it.

closed 2022-06-13 by SP2FET