mudita/MuditaOS #3658: Music app won't resume playback after connecting or disconnecting headset while playback is paused

rawzen, 2022-03-19:
1. Start playing a song in the Music app on Pure with no headset connected.
2. Pause playback.
3. Connect headset. 
4. Press play continue playback.
5. Nothing happens.
6. The app has to be closed and restarted to be able to resume playback.

Same happens if headset is connected at first playback and later disconnected.
If song is kept playing during connect and disconnected everything works fine.

SP2FET, 2022-03-21:
Hi, which firmware version are you using? The latest one (1.1.6)?

rawzen, 2022-03-21:
Yes. It is version 1.1.6.

mån 21 mars 2022 kl. 08:36 skrev Bartosz Cichocki ***@***.***

SP2FET, 2022-03-21:
We've created a ticket and it will be further investigated: MOS-331
Thank you for reporting!

closed 2022-03-28 by Lefucjusz