mudita/MuditaOS #3659: Sometimes long delay when switching to Connected mode

rawzen, 2022-03-19:
When in Disconnected or Do not disturb mode, and switching to Connected it can sometimes take up to a minute to get visual feedback on screen that the mode has changed and network connection is resumed. Other times it only takes a few seconds.
Switching from Disconnected to Do not disturb happens immediately.

SP2FET, 2022-03-21:
Hi, which firmware version are you using? The latest one (1.1.6)?

rawzen, 2022-03-21:
Yes. It is version 1.1.6.

mån 21 mars 2022 kl. 08:36 skrev Bartosz Cichocki ***@***.***

SP2FET, 2022-03-21:
Could you please confirm that after switching the slider to the next position you get a delay between switching the slider and info popup on the screen, or do you mean the delay between switching the connected mode and fully-operational modem?

rawzen, 2022-03-21:
The long delay is between switching the slider to connected mode and the
info popup, and the network connection seems to be down until the info
popup appears on the screen.

mån 21 mars 2022 kl. 12:07 skrev Bartosz Cichocki ***@***.***

SP2FET, 2022-03-21:
alright, thank you for the info, we'll investigate the issue and keep you informed if we found anytihng

Lefucjusz, 2022-08-22:
@rawzen The issue has been fixed and will be present in one of the future releases.

closed 2022-08-22 by Lefucjusz