mudita/MuditaOS #3677: Call reject list

esope, 2022-03-24:
# 🚀 Call reject list (aka black list)

## 📝 Description

I sometimes need to block all incoming calls from a given phone number (like for spams). Having a reject list (or a black list) of such numbers, that the Mudita Pure would automatically block, would be neat.

## 📝 Describe the solution you'd like

Add the possibility to add a phone number or a contact to the reject list.
Add the possibility to remove a phone number from the reject list.
All the phone numbers from the reject list should be blocked for incoming calls.

## 📝 Describe alternatives you've considered

Blocking these numbers manually, every time they call.

pholat, 2022-05-02:
This will be discussed internally by @mjan-dev. It will be followed up in a few days :crossed_fingers:

mjan-dev, 2022-05-09:
Added to our Jira as additional feature, however it is not of high priority. Thanks for adding the sugestion! @esope

mjan-dev, 2022-05-10:
Adding our internal issue link for tracking: [MOS-467](https://appnroll.atlassian.net/browse/MOS-467)

Lefucjusz, 2023-06-12:
Hi @esope, the business decision has been made to not implement this feature. I'm closing this issue now.

closed 2023-06-12 by Lefucjusz