mudita/MuditaOS #3687: Alarms and time change due to daylight saving time

esope, 2022-03-28:
# 🐛 Alarms and time change due to daylight saving time

## 📝 Description

When you set up an alarm and a time change occurs (advance or delay of 1 hour) due to daylight saving time, the alarm does not ring at the expected time, but at the time *before* the time change.

## 👍 Expected behavior

If I set up an alarm at 6AM (winter time), and the daylight saving time occurs during the night, then the alarm should go off at 6AM (summer time), which effectively corresponds to 5AM in winter time.

## 👎 Current behavior

If I set up an alarm at 6AM (winter time), and the daylight saving time occurs during the night, then the alarm will go off at 7AM (summer time), which effectively corresponds to 6AM in winter time.

## 🔬 Minimal reproduction

Set the option "get the time from the network".
Set up an alarm for 6AM the day before the time change occurs.
Wait for the alarm to go off the next morning, after the time change has occurred.

## 🔍 Provide logs

none

## 🌍 Your operating system and MuditaOS version

Linux operating system. MutidaOS 1.1.6.

## 🤔 Anything else relevant?

I remember of a similar bug around 2010 for iPhones.

SP2FET, 2022-03-29:
Hi, thanks for letting us know! Reported as MOS-354

Lefucjusz, 2023-02-06:
Hi @esope! The bug has been resolved and fix will be present in one of the next MuditaOS releases.

closed 2023-02-06 by Lefucjusz