mudita/MuditaOS #3707: Predictive text / T9

GravisZro, 2022-04-03:
# 🚀 Feature request

## 📝 Description
It's been mentioned several times on the forum but as far as I know, predictive text has not been implemented yet.

## 📝 Describe the solution you'd like
Implement real-time text prediction.

These were the best implementations I could find though I think they all need work as they don't seem to support caching the tree structure.

* C implementation of T9: https://github.com/yweweler/c-t9
* C++ version of the first one: https://github.com/yweweler/cpp-t9
* minimalistic C implementation: https://github.com/hy987/T9-Predictive-Text

## 📝 Other info
The patents on the technology have since expired as they are all pre-2000. See https://en.wikipedia.org/wiki/T9_(predictive_text) for details.

SP2FET, 2022-04-04:
Hi, T9 is on our roadmap, and work on this topic will start for sure - we just have to finish ongoing topics. Some initial research has been already made FYI.

GravisZro, 2022-04-04:
Glad to hear it.  I hope the current topics are centered around system issues like baseband modem management because connectivity is still problematic.

TanukiKabuki, 2022-04-18:
This would be a quality improvement for text input.

neutralinsomniac, 2022-05-12:
I wrote an extremely feature-rich (and interactive) T9 engine in Python: https://github.com/neutralinsomniac/t9. My plan is to port the logic to C++ since the logic was the hardest part to get right, but if someone else wants to take a stab at it, feel free.

romangod902, 2022-11-05:
As anyone figured out how to implement this into Mudita Pure. After an extensive search, it only seems there are several developers who have outlined the coding behind T9.  Mudita Forum has threads from years ago, and most recently earlier this year when Mudita Pure developers responded reassuring the "plan to implement" T9 as a "high priority" item.  Several months later without update. Abc/ABC is inefficient no matter how pure/simplistic it may seem. In this case it is subjective -- T9 is more simplistic over Abc/ABC in which case it aligns perfectly with the motto of Mudita. Updates would be appreciated!