mudita/MuditaOS #3708: Modem can become unresponsive

GravisZro, 2022-04-04:
# 🐛 Bug report

## 📝 Description
In some scenarios, the modem can become completely unresponsive and will not respond without cycling the power.

## 👍 Expected behavior

Every command issued to the modem should activate a watchdog timer for triggering the reset line to the modem.

## 👎 Current behavior

The specific scenario I encountered was in the first day of use when I wasn't sure which slot was SIM1 and SIM2 but after inserting the card I never removed it.  I had gotten reception bars earlier in the day but later they were gone and it looked like the SIM icon was X'd out on the right side. Upon trying to switch the SIM slot to use in the settings, I was presented with message `sim_card_not_ready` aka "Waiting for Modem to start. \n\n This may take a moment." which would never expire.  Eventually, I turned the phone off and upon trying to turn it back on, it would not turn on.  The phone was **completely** unresponsive until I removed the battery and replaced it.

## 🔬 Minimal reproduction

Since this is likely due to a bug in the modem, figuring out how to reproduce the bug is going to be difficult to say the least.

## 🌍 Your operating system and MuditaOS version
Model: 1.01
OS Version: 1.1.6

pholat, 2022-04-05:
Hi, would you be so kind to rise this issue via Mudita Center too (and attach your Phone serial number here)
This way we will be able to check the device logs.
Sorry for the inconvenience, we will add the possibility to download logs so that you will be able to attach them here directly in the future.

GravisZro, 2022-04-08:
OK, so I _finally_ found that there were prebuilt ELFs.  It was hard to find a download link on the Mudita website.
My serial number: 27029130741012

GravisZro, 2022-04-17:
I'm closing this since it has no real resolution.

closed 2022-04-17 by GravisZro