mudita/MuditaOS #3726: vCard and iCalendar file generation for USB MTP

GravisZro, 2022-04-09:
# 🚀 Feature request

## 📝 Description

The current USB CDC serial interface is fine but it has limited support: only Mudita Center can interface with Pure Phone.

## 📝 Describe the solution you'd like

I would like the USB MTP interface to include vCard files for contacts and iCalendar files for calendar entries.  Since the actual information is stored in a database on the phone, the files would need to be generated ad hoc.  Doing this would allow you to modify remove and add new entries without using Mudita Center as an extra intermediary, especially using a synchronization program like [vdirsyncer](https://vdirsyncer.pimutils.org/en/stable/).

## 📝 Describe alternatives you've considered

A small FUSE application could do the same thing but it would just be another one-off application.  It would be much nicer if it were built into the phone.

mjan-dev, 2022-04-11:
We are currently not planning to add such functionality, although you could take on the challenge and try to develop it by yourself. We would support you with great pleasure. :smile:

GravisZro, 2022-04-12:
OK, to modify the USB stack's MTP implementation, I will need to be able build firmware and therefore a roadblock for the moment.  I'll close this ticket since the request was declined.

closed 2022-04-12 by GravisZro