mudita/MuditaOS #3729: Backlight shortcut

LizGalanis, 2022-04-10:
# 🚀 Feature request

## 📝 Description

I prefer to use the phone without the backlight, however this means that once it's dark I can't see the screen to turn the light on! This leaves me needing to have the light automatically on at all times. 

## 📝 Describe the solution you'd like

It would be awesome if there was a shortcut on the keypad, or maybe using the torch button, that could turn the back light on and off.

Storms-Engineering, 2022-04-18:
I agree with this one.  Or simply having the backlight only turn on when it is very dark.  Obviously this will require some testing, to see what is the magic value on your photosensor to turn the backlight on.

dr3mro, 2022-07-22:
or add option to make backlight automatic when light sensors detects its dark only

dr3mro, 2022-10-17:
I suggest long press * from home to toggle light on and off

demoive, 2024-08-06:
> or add option to make backlight automatic when light sensors detects its dark only

> I agree with this one. Or simply having the backlight only turn on when it is very dark. Obviously this will require some testing, to see what is the magic value on your photosensor to turn the backlight on.

FYI, as per [this comment](https://github.com/mudita/MuditaOS/issues/4037#issuecomment-1240553905), this feature has already been implemented.

dr3mro, 2024-08-06:
Implemented for pure?