mudita/MuditaOS #3730: Alarms become inactive after reboot

esope, 2022-04-11:
# 🐛 Bug report: Alarms become inactive after reboot

## 📝 Description

After a reboot, the alarm icon of the home screen is not displayed, and the active alarms won't go off.

## 👍 Expected behavior

If some alarm is active, the alarm icon on the home screen should always be displayed, even after a reboot.
The alarms that are marked as active should go off, even after a reboot.

## 👎 Current behavior

After a reboot, if you have set up an alarm before the reboot, the alarm icon on the home screen is not displayed, and your alarm will not go off.

## 🔬 Minimal reproduction

1. Set up an alarm and mark is as active.
2. Go back to home screen: the alarm icon is displayed.
3. Reboot your phone.
4. Activate your SIM card by entering your PIN if necessary.
5. The alarm icon is not displayed (and your alarm will not go off)
6. Go to the alarm settings: you alarm is marked as active. This is inconsistent with the absence of the alarm icon of the home screen. More importantly: although it is marked as active, your alarm will not go off.
7. If you deactivate and reactivate your alarm, and go back to the home screen, then you will see that the alarm icon is back, and your alarm will correctly go off at the right moment.

## 🔍 Provide logs

No logs.

## 🌍 Your operating system and MuditaOS version

MuditaOS 1.2.0
Operating system: Linux (though I don't think this is relevant)

## 🤔 Anything else relevant?

I almost missed an important appointment because of this bug.
This is very annoying, because the phone may reboot by itself without any notice, especially after the 1.2.0 upgrade...

mjan-dev, 2022-04-11:
Sorry to hear about such issue.
Could you please send over issue report through Mudita Center? We would then have access to logs from your device.

esope, 2022-04-11:
> Could you please send over issue report through Mudita Center?

Done.

SP2FET, 2022-07-04:
Hi, we tried to reproduce the bug, but without any success. Could you please let us know if the issue persists?

esope, 2022-07-04:
Yes, it persists.
Something I forgot to put in the bug report: I am using alarms with repetitions (like, "repeat every day of the week").
I have not tested with "one time" alarms.

esope, 2022-07-06:
Also, I do not know whether it is related at all: I configured the "time and date" by disabling "automatic time and date", then setting the right time zone and time, and then re-enabling "automatic time and date".

Lefucjusz, 2022-08-08:
@esope We were able to reproduce the bug, it has been reported under MOS-606 ticket and will be further investigated.

esope, 2022-10-10:
The bug is still there in the 1.4.0 release.

closed 2023-01-16 by Lefucjusz