mudita/MuditaOS #3749: Contact application - UI improvement: search for substring matches

GravisZro, 2022-04-14:
# 🚀 Feature request

## 📝 Description

When searching contacts, the search only matches the beginning of fields.  This means when searching for the first name "Big Jake" that you must search for "big" because "mike" will not match.

## 📝 Describe the solution you'd like

Search for match sub-strings.

## 📝 Describe alternatives you've considered

There could be an "Alias" field for contacts that would be the only substring search if it's a speed issue.

pholat, 2022-05-23:
Hi! it's not on the roadmap right now. Saying that - we will propose this be implemented internally. 
Internal ticket reference: https://appnroll.atlassian.net/browse/MOS-502

Saying that - please feel free to come up with implementation propositions.

Lefucjusz, 2022-07-18:
@GravisZro the issue has been reported under MOS-414 ticket, thank you for your suggestion of the solution.