mudita/MuditaOS #3754: Tea Timer

TanukiKabuki, 2022-04-18:
# 🚀 Feature request

## 📝 Description

Would be nice to have a tea timer on the device, something simple to just choose, white, green, black, herbal, etc. Set the time, set the phone down, wait. Nothing frivolous, just time. I don't have my device yet, but looking at hopping on board and assisting with the development of new functionality.

## 📝 Describe the solution you'd like

**Timers for:**
White Tea
Black Tea
Tisane/Herbal Tea
Green Tea
Oolong Tea

## 📝 Describe alternatives you've considered

Just a standard timer and having to remember what time each type should be.

Have you considered any alternative solutions or workarounds?

pholat, 2022-04-25:
A fine idea, we will propose it to be implemented :)

JohnDumpling, 2022-06-24:
I feel like a universal timer app with the ability to set and rename multiple timers would be more useful.

Lefucjusz, 2023-06-12:
Hi @TanukiKabuki, unfortunately the business decision is to not implement this feature. I'm closing this issue now.

closed 2023-06-12 by Lefucjusz