mudita/MuditaOS #3755: Delete All Calls in Call Log

Storms-Engineering, 2022-04-18:
# 🚀 Feature request
Delete All Calls in Call Log
## 📝 Description
Be able to delete All Calls in Call Log

## 📝 Describe the solution you'd like
You can use the existing options menu, but simply add a "delete all calls" option

## 📝 Describe alternatives you've considered
Deleting every call by hand is kind of annoying.  And i was just thinking that over a couple years the amount of calls might be a lot, and it might slow the phone down.

pholat, 2022-05-02:
Thanks for your feedback! The requested feature was added internally for planning ( Internal link for our reference https://appnroll.atlassian.net/browse/MOS-447 )

mjan-dev, 2022-05-20:
@Storms-Engineering FYI this feature is in progress right now :+1:

Storms-Engineering, 2022-05-23:
@mjan-dev @marziel  Sweet! Thanks guys!

marziel, 2022-05-24:
@Storms-Engineering You're welcome! :)

closed 2022-05-25 by marziel