mudita/MuditaOS #3763: Silent mode

snan, 2022-04-24:
I would want the center position on the right most slider to be silent, but show on the lock screen calls and messages. That's a good complement between airplane and normal, I'd use each about a third of the time each. I was under the impression that it workd that way in 1.1.5 but I'm not sure about that.

patrycja-paczkowska, 2022-04-25:
Hello @snan 
To receive silent notifications on the locked screen after setting the slider to the DO NOT DISTURB position, you should go to MENU -> Settings -> Phone modes -> Do not disturb and set Notification when locked to ON. This way, you should have notifications about incoming calls and messages displayed on your lock screen.

snan, 2022-04-25:
That's what I have, but since 1.2.0 they aren't displayed there until I slide it back up.

patrycja-paczkowska, 2022-04-25:
Can you please confirm that for DO NOT DISTURB mode, this label is displayed on the home screen in the upper left corner?

pholat, 2022-05-23:
@snan would you be so kind to double-check?

pholat, 2022-05-30:
moved to quarantine - with no response we plan to drop it next week.

Lefucjusz, 2022-07-18:
Closing the issue due to lack of response, if the problem persists please reopen the issue.

closed 2022-07-18 by Lefucjusz