mudita/MuditaOS #3769: Music application - UI bug: Presentation sometimes non-functional.

GravisZro, 2022-04-25:
# 🐛 Bug report

## 📝 Description

The music player "presentation" UI sometimes does not show that music is playing and cannot control it.  This specifically happens when I play the 30th song in my list or any after it.

## 👍 Expected behavior

Should play normally.

## 👎 Current behavior

The music player "presentation" UI sometimes does not show that music is playing and cannot control it.  This specifically happens when I play the 30th song or later from the "All songs" menu.

## 🔬 Minimal reproduction

1. Open `MUSIC`->`Music Library`->`All songs`->`the 30th song or later`
2. press `Back` then up to get to the music player "presentation" UI.

## 🌍 Your operating system and MuditaOS version
1.2.0

## 🤔 Anything else relevant?
I loaded a couple hundred songs loaded directly as files via USB MTP.

pholat, 2022-05-02:
Is it ok to just play 30th+ song, or we should wait till it?
What happens after you are not able to control it?

GravisZro, 2022-05-02:
* It only happen upon selection of the 30th+ song.  There is no issue with playing 30+ songs.
* After the the completion of the song playback, the next song will return control to the music player "presentation" UI

patrycja-paczkowska, 2022-05-20:
FYI I was able to reproduce the problem and created a ticket on our internal board https://appnroll.atlassian.net/browse/MOS-496.

Lefucjusz, 2023-02-09:
Hi @GravisZro! The issue has been recently resolved and will be present in one of the next MuditaOS releases.

closed 2023-02-09 by Lefucjusz