mudita/MuditaOS #3779: Alarm crash

snan, 2022-04-27:
Phone crashed when the alarm went of while I was at the alarms list screen.

<small>also had a ton of dropped calls today as per ushe which is off topic to the alarm crash but just a follow up.</small>

pholat, 2022-05-02:
@snan please send the report on Mudita Center so that we would have a crash dump and would be able to reproduce the issue :bow:

snan, 2022-05-02:
I tried and I started Mudita Center, it asked me to update it and after I did that, connecting (between phone & computer) worked once but then it disconnected and this is my life now:
[image]

mjan-dev, 2022-05-06:
Tried to reproduce the alarm issue many times on 1.2.0 but with no success.

pholat, 2022-05-23:
@snan would you be so kind to tell me if:
- your MC application is up to date?
- this problem with communication is persistent?

pholat, 2022-05-23:
edit: we found some similar issue internally ( internal reference: https://github.com/mudita/MuditaOS/issues/3779 )

snan, 2022-05-23:
How do I send logs from MC? Since I've intermittently gotten it to connect again.

mjan-dev, 2022-05-30:
@snan I'm attaching a forum link with a quick description on how to do it. 
https://forum.mudita.com/t/how-to-send-logs-to-mudita-via-mudita-center/4085

Lefucjusz, 2022-06-20:
@snan Have you sent the logs to Mudita via the Mudita Center?

snan, 2022-06-20:
Ah, I didn't see that Urszula had replied in the thread! I'll try again now

Lefucjusz, 2022-10-10:
@snan Please try to update to the latest release - 1.4.0. We've fixed a lot of bugs regarding stability issues since your report, so the OS should be now much stable.

closed 2022-10-11 by Lefucjusz