mudita/MuditaOS #3781: Image assets are 1-bit

GravisZro, 2022-04-28:
# 🐛 Bug report

## 📝 Description

In making my own PGM<->VPI converters (to use with Netpbm to ultimately convert PNG<->VPI), I've found that the source PNG images have been converted to 1-bit images.  Is this intentional?

## 👍 Expected behavior

I expected the source images to be converted to 4-bit grayscale.

## 👎 Current behavior

As far as I can tell, when converting from 24-bit RGB in the PNG the bottom 23-bits are simply discarded.

## 🔬 Minimal reproduction
Open [plus_32px_W_G.png](https://github.com/mudita/MuditaOSPublicAssets/blob/master/source/assets/images/pure/common/plus_32px_W_G.png) and inspect the very top of the `+` symbol.  The black-looking pixels are actually `#0D0D0D`.  If you dump the values for [plus_32px_W_G.vpi](https://github.com/mudita/MuditaOSPublicAssets/blob/master/image/assets/images/plus_32px_W_G.vpi) then you will notice this value is rendered to `00`

## 🔍 Provide logs

Here is a hex dump (with 0F replace with spaces) of the existing VPI image assets (appears to be 1-bit):
```
$ cat plus_32px_W_M.vpi | ./vpitopgm > plus_32px_W_G.pgm
$ hexdump -v -s 12 -e '"%08_ax | " /1 "%02X " /1 "%02X " /1 "%02X " /1 "%02X " /1 "%02X " /1 "%02X " /1 "%02X " /1 "%02X " /1 "%02X " /1 "%02X " /1 "%02X " /1 "%02X " /1 "%02X " /1 "%02X " /1 "%02X " /1 "%02X " /1 "%02X " /1 "%02X " /1 "%02X " /1 "%02X " /1 "%02X " /1 "%02X " /1 "%02X " /1 "%02X " /1 "%02X " /1 "%02X " /1 "%02X " /1 "%02X " /1 "%02X " /1 "%02X " /1 "%02X " /1 "%02X \n"' plus_32px_W_G.pgm | sed -e "s|\(0F\)|  |g"
0000000c |                                                                                                 
0000002c |                                                                                                 
0000004c |                                                                                                 
0000006c |                                                                                                 
0000008c |                                              00 00                                              
000000ac |                                           00 00 00 00                                           
000000cc |                                           00 00 00 00                                           
000000ec |                                           00 00 00 00                                           
0000010c |                                           00 00 00 00                                           
0000012c |                                           00 00 00 00                                           
0000014c |                                           00 00 00 00                                           
0000016c |                                           00 00 00 00                                           
0000018c |                                           00 00 00 00                                           
000001ac |                                           00 00 00 00                                           
000001cc |                00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00                
000001ec |             00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00             
0000020c |             00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00             
0000022c |                00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00                
0000024c |                                           00 00 00 00                                           
0000026c |                                           00 00 00 00                                           
0000028c |                                           00 00 00 00                                           
000002ac |                                           00 00 00 00                                           
000002cc |                                           00 00 00 00                                           
000002ec |                                           00 00 00 00                                           
0000030c |                                           00 00 00 00                                           
0000032c |                                           00 00 00 00                                           
0000034c |                                           00 00 00 00                                           
0000036c |                                              00 00                                              
0000038c |                                                                                                 
000003ac |                                                                                                 
000003cc |                                                                                                 
000003ec |                                                                                                 
```

Here is a hex dump (with 0F replace with spaces) of 4-bit images translated from the PNG source file:
```
$ cat plus_32px_W_G.png | pngtopam -mix -background=white | pnmremap -quiet -nofs -mapfile=palette.pgm > plus_32px_W_G.pgm 
$ hexdump -v -s 12 -e '"%08_ax | " /1 "%02X " /1 "%02X " /1 "%02X " /1 "%02X " /1 "%02X " /1 "%02X " /1 "%02X " /1 "%02X " /1 "%02X " /1 "%02X " /1 "%02X " /1 "%02X " /1 "%02X " /1 "%02X " /1 "%02X " /1 "%02X " /1 "%02X " /1 "%02X " /1 "%02X " /1 "%02X " /1 "%02X " /1 "%02X " /1 "%02X " /1 "%02X " /1 "%02X " /1 "%02X " /1 "%02X " /1 "%02X " /1 "%02X " /1 "%02X " /1 "%02X " /1 "%02X \n"' plus_32px_W_G.pgm | sed -e "s|\(0F\)|  |g"
0000000c |                                                                                                 
0000002c |                                                                                                 
0000004c |                                                                                                 
0000006c |                                                                                                 
0000008c |                                           0B 01 01 0B                                           
000000ac |                                           07 00 00 07                                           
000000cc |                                           07 00 00 07                                           
000000ec |                                           07 00 00 07                                           
0000010c |                                           07 00 00 07                                           
0000012c |                                           07 00 00 07                                           
0000014c |                                           07 00 00 07                                           
0000016c |                                           07 00 00 07                                           
0000018c |                                           07 00 00 07                                           
000001ac |                                           07 00 00 07                                           
000001cc |             0B 07 07 07 07 07 07 07 07 07 04 00 00 04 07 07 07 07 07 07 07 07 07 0B             
000001ec |             01 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 01             
0000020c |             01 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 01             
0000022c |             0B 07 07 07 07 07 07 07 07 07 04 00 00 04 07 07 07 07 07 07 07 07 07 0B             
0000024c |                                           07 00 00 07                                           
0000026c |                                           07 00 00 07                                           
0000028c |                                           07 00 00 07                                           
000002ac |                                           07 00 00 07                                           
000002cc |                                           07 00 00 07                                           
000002ec |                                           07 00 00 07                                           
0000030c |                                           07 00 00 07                                           
0000032c |                                           07 00 00 07                                           
[... cut ...]
0000036c |                                           0B 01 01 0B                                           
0000038c |                                                                                                 
000003ac |                                                                                                 
000003cc |                                                                                                 
000003ec |                                                                                                 
```

pholat, 2022-05-02:
@GravisZro we do not plan to support your tools. It might be possible that we will release our changed converter for public use with documentation. Though this has to be discussed in the company internally. 

Unfortunately at the moment if you really wish to create tooling - it will require you to actually analyze how it works as it has minimal to no documentaion.

closed 2022-05-02 by pholat

GravisZro, 2022-05-02:
Please reopen this issue because this is not about the tools I made, I just used them to make the issue I discovered clear.  I don't care if the tools are rejected, I am trying to point out that the tools used to generate the VPI files are discarding details from the source PNG images.  If you read this post, I will walk through each step to demonstrate this fact.

Here is the step-by-step process used to dump PNG pixel values using nothing but normal tools:
* The first two lines set up P as the `hexdump` output format string.  It's just a 16-bit address followed by 32 groups of 3-byte RGB data (i.e. "xxxx | RRGGBB (repeat RRGGBB 31 more times)").
* The PNG file, `plus_32px_W_G.png` is printed to standard output
* `pngtopam` reads the PNG stream and prints out the image as a binary 24-bit [Portable PixMap image](https://en.wikipedia.org/wiki/Netpbm#PAM_graphics_format).  No colors are manipulated in any way, they are written as is.  (Note: If you add `-plain` to the command it will print out as an ASCII 24-bit Portable PixMap image which is a very human readable format.)
* `tail` only takes the last line to remove the header data which states the dimensions and image type
* `hexdump` convert to the binary data stream to ASCII, printing in a format specified in P
* The very wide hex dump (232 characters wide) processed with `sed` with 3 regular expression to make the output readable.
  * Capture two hex digits (RR) by four other hex digits.  Replace all six with just the RR hex digits since it's really an grayscale image.
  * Replace all instances of "FF" with spaces (don't clutter output with white)
  * This last expression is used to color the 0D characters (which didn't copy over)
```
MuditaOSPublicAssets$ P='/1 "%02X"/1 "%02X"/1 "%02X "' \
> P='"%04_ax | "'$P$P$P$P$P$P$P$P$P$P$P$P$P$P$P$P$P$P$P$P$P$P$P$P$P$P$P$P$P$P$P$P'"\n"'; \
> cat source/assets/images/pure/common/plus_32px_W_G.png | \
> pngtopam | \
> tail -n 1 | \
> hexdump -v -e "$P" | \
> sed -e "s|\([[:xdigit:]]\{2\}\)[[:xdigit:]]\{4\}|\1|g" \
>     -e "s|\(FF\)|  |g" \
>     -e "s|0D|$(tput setaf 1)&$(tput sgr0)|g"
0000 |                                                                                                 
0060 |                                                                                                 
00c0 |                                                                                                 
0120 |                                                                                                 
0180 |                                           BD 0D 0D BD                                           
01e0 |                                           7F 00 00 7F                                           
0240 |                                           7F 00 00 7F                                           
02a0 |                                           7F 00 00 7F                                           
0300 |                                           7F 00 00 7F                                           
0360 |                                           7F 00 00 7F                                           
03c0 |                                           7F 00 00 7F                                           
0420 |                                           7F 00 00 7F                                           
0480 |                                           7F 00 00 7F                                           
04e0 |                                           7F 00 00 7F                                           
0540 |             BD 7F 7F 7F 7F 7F 7F 7F 7F 7F 3F 00 00 3F 7F 7F 7F 7F 7F 7F 7F 7F 7F BD             
05a0 |             0D 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 0D             
0600 |             0D 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 0D             
0660 |             BD 7F 7F 7F 7F 7F 7F 7F 7F 7F 3F 00 00 3F 7F 7F 7F 7F 7F 7F 7F 7F 7F BD             
06c0 |                                           7F 00 00 7F                                           
0720 |                                           7F 00 00 7F                                           
0780 |                                           7F 00 00 7F                                           
07e0 |                                           7F 00 00 7F                                           
0840 |                                           7F 00 00 7F                                           
08a0 |                                           7F 00 00 7F                                           
0900 |                                           7F 00 00 7F                                           
0960 |                                           7F 00 00 7F                                           
09c0 |                                           7F 00 00 7F                                           
0a20 |                                           BD 0D 0D BD                                           
0a80 |                                                                                                 
0ae0 |                                                                                                 
0b40 |                                                                                                 
0ba0 |                                                                                                 
```
[... cut ...]
    * `0x0027` - two bytes: vector offset: 0
    * `0x0029` - one byte: vector run-length: 2 pixels (0x02)
    * `0x002A` - one byte: vector color: `0x00` **(Detail lost!  Expected 0x01)**
  * vector 3
    * `0x002B` - two bytes: vector offset: 0
    * `0x002D` - one byte: vector run-length: 1 pixel
    * `0x002E` - one byte: vector color: `0x0B`

So manually decoding the data shows that the pixels that should be `0B 01 01 0B` are actually `0B 00 00 0B`.  This is the loss of detail that I've been trying to explain.