mudita/MuditaOS #3789: Extract multimedia metadata upon transfer

GravisZro, 2022-04-28:
# 🚀 Feature request

## 📝 Description

Currently when you transfer songs, you need a client that will specify the metadata (title, artist, etc) despite the metadata being in the file itself.

## 📝 Describe the solution you'd like

I would like the multimedia information to be extracted from the FLAC/MP3 file **if** the information is _not_ supplied after transferring to the phone.

## 📝 Describe alternatives you've considered

Making a robust native MTP client would be great but that's it's own can of worms.

pholat, 2022-05-02:
Can you please elaborate on where is the issue? Is it when using the Mudita Centre client? Or in phone runtime?

GravisZro, 2022-05-02:
It should work with any client, so it's an issue with the phone runtime.

pholat, 2022-05-23:
Ok, could you be kind enough to elaborate a bit more? Currently, we have an issue properly understanding where is the issue.

GravisZro, 2022-05-23:
Currently, when transferring songs via MTP, I transfer them as files, not as songs which would include metadata like the artist, album, song title, etc. which (as far as I can tell) means that these songs are not entered into the Pure's media database.  Perhaps Windows transfers via MTP as songs automatically but I do not have Windows, only Linux.  I believe they are not in Pure's database because in the player I can _only_ access my songs via the "All Songs" menu, since the "Artists", "Albums", and "Playlists" menus are grayed out.  However, in the "All Songs" menu, they appear as the artist and song title which implies the metadata is being extracted at some point.

pholat, 2022-05-30:
I'm afraid this doesn't really work how you think it works. @SP2FET please provide full description and explanation

SP2FET, 2022-05-30:
@GravisZro let me explain:
- menu entries that are greyed out because those are not implemented yet - it does not rely on files you have on the phone.
- transferring files via MTP does not alter any metadata of the file - tags are included into file and are copied to the phone - you can verify it by checking the title and artist while playing the audio. If there were no tags, you wouldn't see the title and artist, only the filename.

Hope it explains everything. We have implementation of the missing entires in our backlog FYI.
Please close the issue if you don't have any more question in this topic ;)

closed 2022-05-30 by GravisZro