mudita/MuditaOS #3795: Messages should be sent when possible

GravisZro, 2022-04-30:
# 🚀 Feature request

## 📝 Description

When sending a text messages fails, it should try again when signal strength increases.

## 📝 Describe the solution you'd like

When it finally does send, instead of it just the time sent by your messages (pressed "send"), it should also include "delayed: %time actually sent%" which could be different day, so keep that in mind.

If there is a reason this isn't being done by default then the ability to turn on this feature would be nice.

## 📝 Describe alternatives you've considered

A notification saying a message failed to be sent wouldn't be as good but it's better than nothing.

pholat, 2022-05-02:
We are not planning such a feature right now. Not sent messages are marked as not sent. This is not as simple as you think - as you are using a currently active sim card and you would need to have this into consideration when implementing it. 
Therefore we will present it, but please do not mind if it won't be selected as a proposition to implement.

Lefucjusz, 2022-07-18:
Currently we're not planning to add such functionality, you can resend the message that failed to send by focusing on it, pressing "Options" and "Send again".

closed 2022-07-18 by Lefucjusz