mudita/MuditaOS #3798: Messages UI: long messages appear truncated

GravisZro, 2022-05-01:
# 🐛 Bug report

## 📝 Description

After receiving a rather long text message, I thought it was truncated because the message just ends.  There is no indication that there is more to the message.

## 👍 Expected behavior

There should be some or of indicator that there is more to the message, like a scroll bar on the side or a downward pointing arrow at the bottom when there is more text below and a upward pointing arrow at the top when there is more text above.

It would be better to fully expand the "bubble" to fit the entire message if possible, to the whole screen if needed.  If it still doesn't fit then being able to select the "bubble" to read it would better because it would still allow you to go through the message history without pressing up and down several extra times for each long message (inconsistent UI behavior is bad ☹️).

## 👎 Current behavior

Long messages just end.  There is no indication that there is more to the message.

## 🔬 Minimal reproduction

Receive a very long text message.

## 🤔 Anything else relevant?
The scrolling behavior is poor because the "bubble" width is recalculated based on which part of the message text is displayed.  This means it resized with every single scroll which is exceptionally distracting when trying to read.

pholat, 2022-05-02:
@GravisZro unfortunately this wonky behavior is by design, the dev team couldn't enforce the scroll bar as it wasn't pretty :) 
Saying that - I'll address this issue with our PO as we totally agree that this is not how it should look like and should be fixed.

Our internal issue for reference: https://appnroll.atlassian.net/browse/MOS-446

Lefucjusz, 2022-09-19:
@GravisZro it was decided that the behavior is not very intuitive indeed, so it will be changed in one of the future releases of MuditaOS. The internal ticket for this issue: https://appnroll.atlassian.net/browse/MOS-653