mudita/MuditaOS #3815: Better thethering support

pisajew, 2022-05-08:
# 🚀 Feature request

## 📝 Description

I'm currently facing issues when trying to use Mudita as a modem on OpenBSD system. When plugged in, it gets detected as:

```
umodem0 at uhub0 port 1 configuration 1 interface 1 "Mudita Mudita Pure" rev 2.00/1.01 addr 2
umodem0: data interface 2, has no CM over data, has break
umodem0: status change notification available
ucom0 at umodem0
ugen0 at uhub0 port 1 configuration 1 "Mudita Mudita Pure" rev 2.00/1.01 addr 2
```

but looks it does not implement hayes AT command set (attached serial port is reported by minicom does not respond to AT commands).

When I "enable thethering", the following happens:

```
ucom0 detached
umodem0 detached
ugen0 detached
umb0 at uhub0 port 1 "Quectel EG25-G" rev 2.00/3.18 addr 2
umb0: missing MBIM descriptor
```

which probably means that while Mudita identifies itself as MBIM-compliant device, its USB interface does not fully implement MBIM standard.

## 📝 Describe the solution you'd like

Either implement missing USB descriptors to make MBIM thehering mode usable or expose Hayes-compatible serial interface which would allow establishing ppp connection.

## 📝 Describe alternatives you've considered

no workaround besides putting Mudita into the drawer and switching back to Android phone.

SP2FET, 2022-05-09:
Hi, during tethering, we connect the USB port of the Quectel EG25-G modem to Pure's modem - Pure acts then as a passthrough, so it seems that your system has a problem with the modem itself.

pisajew, 2022-05-09:
Thanks. Will doublecheck the OS side then.

closed 2022-05-09 by pisajew