mudita/MuditaOS #3816: Messages and calls not coming through

LizGalanis, 2022-05-08:
# 🐛 Bug report

## 📝 Description

A clear and concise description of the problem...

## 👍 Expected behavior

What you think should happen...

## 👎 Current behavior

Current 'buggy' behavior...

## 🔬 Minimal reproduction

Simple steps to reproduce this bug.

Issues that don't have enough info and can't be reproduced will be closed.

## 🔍 Provide logs

Please paste relevant logs below or attach them to this issue.

<pre><code>

</code></pre>

## 🌍 Your operating system and MuditaOS version

## 🤔 Anything else relevant?

mjan-dev, 2022-05-20:
Hi @LizGalanis. Thanks for reporting. We have some idea of what could cause this problem and are working on a possible fix. Would it be possible to send a report through Mudita Center so we can make sure this is the same issue?

LizGalanis, 2022-05-20:
Hi Mjan, can you let me know the steps for sending a report through Mudita Centre?

pholat, 2022-05-24:
You can check the relevant how-to on our forum here: https://forum.mudita.com/t/how-to-send-logs-to-mudita-via-mudita-center/4085

Lefucjusz, 2022-06-20:
@[LizGalanis](https://github.com/LizGalanis) Have you sent the logs to Mudita via Mudita Center?

LizGalanis, 2022-06-28:
Hi Lefucjusz, I have sent one now. It's a while since I've used the phone though so I'm not sure if it's helpful? I had to stop using the phone since I could not rely on it for calls, texts or alarm clock.

Lefucjusz, 2022-07-18:
@LizGalanis The issue has been reported under MOS-395 ticket.

Lefucjusz, 2022-11-28:
Hi @LizGalanis, sorry for the late update - the issue has been fixed in 1.4.0 MuditaOS version. Closing this thread now.

closed 2022-11-28 by Lefucjusz