mudita/MuditaOS #3823: Phone unresponsible before alarm clock time

pisajew, 2022-05-11:
# 🐛 Bug report

## 📝 Description

Phone unresponsive when alarm clock should fire. Constantly vibrating, emitting weird noises.
## 👍 Expected behavior

What you think should happen... And what do YOU think shoud happen?

## 👎 Current behavior

I left my phone connected with usb cable to a PC for charging. Keyboard was disabled and phone mode was set to do not disturb. When alarm time came the phone activated vibration motor and started emmitting noises (like playing back the alarm music but without meeting timing deadline for filling playback buffer maybe because of highly overloaded CPU). The keyboard is unresponsive. Display stucked at clock showing alarm time minus one minute. The phone is not responding to the keyboard, I see no obivous way to restart it / turin it off. Eventually, 1..2 minutes after I removed SIM card, the phone did turn off the buzzer and stopped making sounds. The keyboard was still unresponsive. When I tried to remove battery it finally rebooted. The whole thing was like 12 minutes.

## 🔬 Minimal reproduction

Simple steps to reproduce this bug.

Issues that don't have enough info and can't be reproduced will be closed. Yea. Go ahead and close it.

## 🔍 Provide logs

Please paste relevant logs below or attach them to this issue.

<pre><code>

</code></pre>

## 🌍 Your operating system and MuditaOS version
the most recent one
## 🤔 Anything else relevant?

SP2FET, 2022-05-16:
Could you please fill up the template as it's supposed to be? And if it's possible, please send logs via Mudita Center. This template has its purpose - we are not able to reproduce some bugs when the description is not complete.
Saying so, once again, please send logs via Mudita Center and paste here your serial number to be able to distinguish your logs from the rest. We'll try to reproduce the bug on our own either way.

esope, 2022-05-16:
I experienced the same bug.
It happens (sometimes) when the phone is charging and (I guess) when it remains plugged for a long time while the battery is fully charged (like when you plug the phone before getting to sleep, and the bug happens the next morning).
The sound of the alarm stutters, as if the CPU was overloaded. Some process getting crazy, or some infinte loop, maybe ?
This bug is very annoying.

adamgas, 2022-05-20:
same - after second snooze, the third alarm didn't went off normally - it just buzzed constantly, no response from keyboard clicks, self-resetted after few minutes

pholat, 2022-05-23:
@esope @adamgas thanks! this is the start we could work with!
Internal bug reference: https://appnroll.atlassian.net/browse/MOS-503
@pisajew would you be so kind as to properly fill in the issue?

Maciej-Mudita, 2022-06-24:
@pisajew @esope @adamgas the problem will be fixed in the new version 1.3.0

closed 2022-06-24 by Maciej-Mudita