mudita/MuditaOS #3828: MuditaOS should not alter contents of received text message

pisajew, 2022-05-11:
# 🐛 Bug report

## 📝 Description

MuditaOS annoyingly inserts hyphenation characters ('-') into text of received SMS, whenever it finds that it should wrap the text to another line.

This raises problems with messages which bear i.e. authentication codes, where one cannot reason from context, what the original text was.

The message is presented correctly in mudita center.

## 👍 Expected behavior

Ideally muditaOS should present text message as it has been received, without adding any additional characters.

## 👎 Current behavior

MuditaOS inserts dash / hyphenation character whenever it needs to wrap a "word" to another line.

## 🔬 Minimal reproduction

Send the following text message "01234567890qwertyuiopasdfghjkl" to a MuditaOS phone (quotes should be stripped when sending).

Mudita center will present this message exactly as sent (i.e.: 01234567890qwertyuiopasdfghjkl)

On phone, however this will be presented as:
01234567890qwertyuio-
pasdfghjkl

So for the receiver it looks like the dash character (-) is a part of original message. While this might look nice for natural language text it makes SMS nearly unusable as medium for transfer of one time passwords etc.

## 🔍 Provide logs

Please paste relevant logs below or attach them to this issue.

<pre><code>

</code></pre>

## 🌍 Your operating system and MuditaOS version
1.2.0
## 🤔 Anything else relevant?

My phone's locale is set to Polish

SP2FET, 2022-05-16:
Hi, indeed, it should be rewritten to provide proper handling of e.g. passwords. We'll take a look at this.

SP2FET, 2022-06-13:
After research and talks with our team, we decided that this is a very specific corner case; thus we can't alter our displaying algorithm to suit this case.

closed 2022-06-13 by SP2FET