mudita/MuditaOS #3840: Mudita sometimes stucked on "Turn off the phone?"

gabryatfendor, 2022-05-16:
# 🐛 Bug report

## 📝 Description

When checking the phone it's stucked at the screen "Turn off the phone?" (I don't know if I accidentally pressed the turn off key while in my pocket) and I can't do anything until the auto-lock activates and, from there on, everything works normally. During the freeze the only button that works is the switch between mode.

## 👍 Expected behavior

Mudita OS should not be freezed

## 👎 Current behavior
Mudita OS is freezed until the auto-lock activates

## 🔬 Minimal reproduction

I don't have really a step-by-step to reproduce, since I noticed this behavior twice in a couple of days when checking my phone

## 🔍 Provide logs

## 🌍 Your operating system and MuditaOS version

Mudita OS 1.2.0

## 🤔 Anything else relevant?

mjan-dev, 2022-05-20:
Hi @gabryatfendor! Thanks for reporting. Would it be possible to send a report through Mudita Center? We would be able to look at logs from your device as we haven't observed such behavior yet.

gabryatfendor, 2022-05-20:
I have enabled the automatic send of the logs from the Mudita Center already. Do I have to send them manually in any other way?

SP2FET, 2022-06-13:
Hi, Sorry for the late response.
The issue indeed exists. we are able to somehow reproduce it. It's already reported under the MOS-210 ticket.

SP2FET, 2022-07-04:
Hi, FYI it's still in progress

Lefucjusz, 2022-07-18:
The issue has been solved and will be fixed in the next MuditaOS release.

closed 2022-07-18 by Lefucjusz