mudita/MuditaOS #3854: SMS merged with old one

gabryatfendor, 2022-05-22:
# 🐛 Bug report

## 📝 Description

I'll try to be as specific as possible. A couple of days ago I received an sms from a contact containing this exact text "a ...". The day after (I received other sms normally meanwhile) another sms from another contact arrived and it was formatted in this way: 
Old message from the other contact that I did not received completely the day before (trailing "a ..." excluded) + part of this new message. Basically it was like the old message I didn't receive had overwritten the new one. I asked, for the sake of curiosity, my contact to send the new message again and this time I received it normally
## 👍 Expected behavior

SMS should not merge

## 👎 Current behavior

It's like the old SMS was in some sort of cache (in the phone? I don't know) and was randomly inserted on the top of this new one

## 🔬 Minimal reproduction

It was completely random so I don't know how to reproduce here

## 🔍 Provide logs

## 🌍 Your operating system and MuditaOS version

Mudita OS 1.2.0

## 🤔 Anything else relevant?

pholat, 2022-05-23:
@gabryatfendor thanks for the report, was it a singular accident, or more frequent behavior?
Saying that: internal issue rose:https://appnroll.atlassian.net/browse/MOS-504

gabryatfendor, 2022-05-23:
@pholat it happened just this time. If you need the complete content of the messages let me know and I will share it privately

esope, 2022-05-26:
I experienced this bug 3 or 4 times last week. Quite annoying.

pholat, 2022-05-30:
@gabryatfendor I expect that the issue is not related to the messages content, but rather specific situation, and wanted to estimate how hard will it be to reproduce it :) Thanks for the report!

christsch, 2022-07-26:
I had the same Problem with MuditaOS 1.2.1:
This happens if I receive a message with more than 160 characters!

esope, 2022-09-09:
I have received a very long SMS, that interfered with itself!
On the phone, I see it scrambled with itself (some chunks of the message are repeated, and then the next chunk is displayed).
On Mudita Center, however, it is displayed perfectly normally.
So I guess the message is correctly stored on the device, but it is displayed incorrectly...

tobald, 2022-09-12:
I experienced the same issue.

dawidwojtas-mudita, 2022-09-14:
@tobald, @esope Can you send logs using Mudita Center? It'll be easier to investigate the issue. Thanks!

tobald, 2022-09-14:
@dawidwojtas-mudita : In Settings>General I clicked "Send mudita center logs to mudita", is this what you mean?

The current bug description is correct. Long SMS do not arrive fully, subsequent content seems to originate from a different contact, at some point here SMS seems completely messed (wrong contact and/or message), as if sms content queue is growing/accumulating at phone company provider.

tobald, 2022-09-14:
> Does it happen always when SMS message is longer than 160 characters?

I don't know, It seems to me the issue starts with a long message but stays present afterwards.

I sent the log, thanks.

dawidwojtas-mudita, 2022-09-15:
@tobald Thanks for the logs and explanation. Probably I've found the issue. The issue will be fixed in the 1.4.0 release.

Lefucjusz, 2022-10-03:
The bug has been fixed and will be published in MuditaOS 1.4.0 release.