mudita/MuditaOS #3855: MMS are not hadled adequately

rumpole, 2022-05-23:
# 🐛 Bug report

## 📝 Description

When someone sends you an MMS to the Mudita Pure, it shows the sender as simply being from "C" with no number or contact detail, and the content is a series of unicode boxes. 

## 👍 Expected behavior

Either one of options i would like to see implemented as the expected behaviour:
1) Implement MMS into the code.  We know the Quectel EC25 module (what the Pure uses right ?) itself can receive MMS, as others online have [documented it working](https://forums.quectel.com/t/eg25-timeout-on-at-qmmsend-and-qping/4780).
2) at the very least, read the Sender address phone number and display that correctly, and also provide some dummy text advising the recipient that they've received an MMS but it is unsupported.  Perhaps with an option to 'quick-reply' to the sender advising them we don't receive MMS and to email the message instead.

## 👎 Current behavior

1) Sender contact name / phone number simply shows up as coming from "C"
2) MMS content simply shows up with many unicode boxes.

## 🔬 Minimal reproduction

Have someone MMS your Pure handsets number and you'll see the issue right away.

## 🔍 Provide logs

Please paste relevant logs below or attach them to this issue.

<pre><code>
I have uploaded logs via the Mudita Ceter, however i believe this problem is well known / widespread already.
</code></pre>

## 🌍 Your operating system and MuditaOS version
1.2.0
## 🤔 Anything else relevant?

pholat, 2022-05-30:
We currently do not have MMS handling implemented. At the moment we do not know whether the message is an SMS or MMS - which will require some major changes in SMS handling. We have this topic on the backlog, though it might take some time due to the current priorities.

awulkiew, 2022-08-23:
Fix: https://github.com/mudita/MuditaOS/pull/4017

closed 2022-08-26 by awulkiew